maniacbug/StandardCplusplus #12: StandardCplusplus Library on Arduino Due

CobrAlex, 2015-03-30:
StandardCplusplus library copy & paste to Arduino/libraries/ and write sketch..

[code]
#include <StandardCplusplus.h>
...
[/code]

but i got error message :

C:\Users\Alex\AppData\Local\Temp\build6039888874897731639.tmp/core.a(new.cpp.o): In function `operator new[](unsigned int)':
C:\Users\Alex\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.2\cores\arduino/new.cpp:26: multiple definition of `operator new[](unsigned int)'
StandardCplusplus\new_opv.cpp.o:C:\Users\Alex\Documents\Arduino\libraries\StandardCplusplus/new_opv.cpp:24: first defined here
C:\Users\Alex\AppData\Local\Temp\build6039888874897731639.tmp/core.a(new.cpp.o): In function `operator delete[](void*)':
C:\Users\Alex\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.2\cores\arduino/new.cpp:34: multiple definition of `operator delete[](void*)'
StandardCplusplus\del_opv.cpp.o:C:\Users\Alex\Documents\Arduino\libraries\StandardCplusplus/del_opv.cpp:25: first defined here
collect2.exe: error: ld returned 1 exit status

Anyone have idea?

mjs513, 2015-04-23:
Getting the same typr errors on an Arduino Mega 2560 using the provided example.  Have no idea.  Did you every get this resolved?  I am running Arduino IDE 1.6.1.

zallesov, 2015-04-23:
Having the same kind of trouble with DUE, while for UNO it compiles just fine

My error message looks different.
```
Applications/Arduino.app/Contents/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -std=gnu++11 -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=10601 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Due" -I/Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/libsam -I/Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/CMSIS/CMSIS/Include/ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/sam/system/CMSIS/Device/ATMEL/ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/sam/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/sam/variants/arduino_due_x -I/Users/alexanderzallesov1/Arduino/libraries/StandardCplusplus -I/Users/alexanderzallesov1/Arduino/libraries/Adafruit_NeoPixel /var/folders/tx/31wk0fb91fj3jjtd7bq_xc0h0000gq/T/build2038817191093835371.tmp/magic_board.cpp -o /var/folders/tx/31wk0fb91fj3jjtd7bq_xc0h0000gq/T/build2038817191093835371.tmp/magic_board.cpp.o 
In file included from /var/folders/tx/31wk0fb91fj3jjtd7bq_xc0h0000gq/T/build2038817191093835371.tmp/color_gradient.h:3:0,
                 from /var/folders/tx/31wk0fb91fj3jjtd7bq_xc0h0000gq/T/build2038817191093835371.tmp/palette_gradient.h:2,
                 from /var/folders/tx/31wk0fb91fj3jjtd7bq_xc0h0000gq/T/build2038817191093835371.tmp/factory.h:5,
                 from magic_board.ino:4:
/Users/alexanderzallesov1/Arduino/libraries/StandardCplusplus/unwind-cxx.h:176:36: error: ISO C++ forbids casting to an array type '_Unwind_Exception_Class {aka char [8]}'
 = ((((((((_Unwind_Exception_Class) 'G' 
                                    ^
/Users/alexanderzallesov1/Arduino/libraries/StandardCplusplus/unwind-cxx.h:177:36: error: ISO C++ forbids casting to an array type '_Unwind_Exception_Class {aka char [8]}'
   << 8 | (_Unwind_Exception_Class) 'N')
                                    ^
/Users/alexanderzallesov1/Arduino/libraries/StandardCplusplus/unwind-cxx.h:178:35: error: ISO C++ forbids casting to an array type '_Unwind_Exception_Class {aka char [8]}'
  << 8 | (_Unwind_Exception_Class) 'U')
                                   ^
/Users/alexanderzallesov1/Arduino/libraries/StandardCplusplus/unwind-cxx.h:179:41: error: ISO C++ forbids casting to an array type '_Unwind_Exception_Class {aka char [8]}'
        << 8 | (_Unwind_Exception_Class) 'C')
                                         ^
/Users/alexanderzallesov1/Arduino/libraries/StandardCplusplus/unwind-cxx.h:180:40: error: ISO C++ forbids casting to an array type '_Unwind_Exception_Class {aka char [8]}'
       << 8 | (_Unwind_Exception_Class) 'C')
                                        ^
/Users/alexanderzallesov1/Arduino/libraries/StandardCplusplus/unwind-cxx.h:181:39: error: ISO C++ forbids casting to an array type '_Unwind_Exception_Class {aka char [8]}'
      << 8 | (_Unwind_Exception_Class) '+')
                                       ^
/Users/alexanderzallesov1/Arduino/libraries/StandardCplusplus/unwind-cxx.h:182:38: error: ISO C++ forbids casting to an array type '_Unwind_Exception_Class {aka char [8]}'
     << 8 | (_Unwind_Exception_Class) '+')
                                      ^
/Users/alexanderzallesov1/Arduino/libraries/StandardCplusplus/unwind-cxx.h:183:37: error: ISO C++ forbids casting to an array type '_Unwind_Exception_Class {aka char [8]}'
    << 8 | (_Unwind_Exception_Class) '\0');
                                     ^
Error compiling.
```

maniacbug, 2023-07-05:
Should be fixed now in 41b4465634e34fce8e010386bd80f9a371c6b311. Working great on my Due.

closed 2023-07-05 by maniacbug