maniacbug/StandardCplusplus #13: queue (and others) interrupt safe?

vyruz1986, 2015-05-25:
Hi,

I am trying to use this library to have a FIFO buffer for ISR's.
In the ISR I am calling the `push()` member function of a `std::queue` object like so:
```c++
struct STriggerRecord
{
   volatile int iSensorNumber;
   volatile int iSensorState;
   volatile unsigned long lTriggerTime;
};
std::queue<STriggerRecord> _STriggerQueue;
void RaceHandlerClass::TriggerSensor2()
{
   STriggerRecord S2Trigger;
   S2Trigger.lTriggerTime = micros();
   S2Trigger.iSensorNumber = 2;
   S2Trigger.iSensorState = digitalRead(_iS2Pin);;
   _STriggerQueue.push(S2Trigger);
}
```

However the last line of this code causes my arduino to freeze.
After I posted this question on arduino.stackexchange, I got [a comment from someone][1] who implies the freeze is caused by 'malloc() not being reentrant`, and since the `.push()` member function of a queue tries to allocate memory, it causes the freeze.

Is there any way around this? I really need a FIFO buffer for my ISR's, and wouldn't know how to achieve this other than using an `std::queue` container (or similar).

[1]: http://arduino.stackexchange.com/questions/12127/crash-when-using-stdqueue?noredirect=1#comment21632_12127

rajuashok, 2016-08-20:
Did you figure this out? I'm trying to use a queue as well for me interrupt

mike-matera, 2016-08-20:
Hi, 

You can't use a queue this way in Arduino because you should never call malloc() in an ISR. Also, you can't use a mutex because an ISR should never block. Take a look at the documentation for Arduino's attachInterrupt():

https://www.arduino.cc/en/Reference/AttachInterrupt

That shows an ISR setting a variable declared 'volatile' and the main loop checking that variable. Your implementation must follow that pattern.

closed 2023-07-05 by maniacbug