maniacbug/StandardCplusplus #14: functional not functional

Makuna, 2015-06-11:
using the latest Arduino IDE, I try to define a callback type that will be lambda compatible like...
```
typedef std::function<void(float progress)> AnimUpdateCallback;
```
and I get a compile error about std.

ziqin, 2016-01-04:
@Makuna

Seems there is no `std::function` template in the `functional` file of this project.

As Vlpp library implement `vl::Func` (which acts like `std::function`) without relying on STL, it may be of use to you. I've ported it to Arduino: <https://github.com/ziqin/ArdComLib/tree/lambda/Vlpp>

marcusrugger, 2016-04-10:
@ziqin 

Thanks for that link.  Very helpful.  I created a repo with just Vlpp and set it up as a proper Adruino library.  You can now pull it down as a git submodule under the libraries folder in your project.

[functional-vlpp repo](https://github.com/marcusrugger/functional-vlpp)

I also wanted to point out that I discovered a bug in the Adruino port.  vl::Func does reference counting on the object it holds.  Instead of increment/decrementing the counter, the port was increment/decrementing the pointer to the counter, causing bizarre, hard to identify problems.  I have fixed it in my repo.

Since my repo isn't a clone of yours, I don't think using a pull request will work well.  If you're interested in the fix, please take a look at this [commit](https://github.com/marcusrugger/functional-vlpp/commit/140a07b6e166a4cf2e15f03b98f6c74f62b90aab).

maniacbug, 2023-07-05:
Sorry, functional objects came in C+11, and are not here in this library.

closed 2023-07-05 by maniacbug

Makuna, 2023-07-05:
@maniacbug What?  I don't understand your comment.
Std::Function objects are part of STL, not c++.  Lamda support is part c++.  They are two different things.  Using Lamda is improved/made easier using functional.  But as stated above, it is missing from "functional" header file (part of stl).