maniacbug/StandardCplusplus #15: Support ESP8622 platform

probonopd, 2015-06-30:
It would be great if the ESP8266 platform could be supported.
https://github.com/esp8266/Arduino

Currently compilation stops and gives errors like
```
/Users/me/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/me/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-804-g2d340c7/tools/sdk//include -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_MOD_WIFI_ESP8266 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/me/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-804-g2d340c7/cores/esp8266 -I/Users/me/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-804-g2d340c7/variants/generic -I/Users/me/Documents/Arduino/libraries/StandardCplusplus-master -I/Users/me/Documents/Arduino/libraries/StandardCplusplus-master/utility /Users/me/Documents/Arduino/libraries/StandardCplusplus-master/complex.cpp -o /var/folders/x3/6g2s9w_j5h194ry3ssgdd56c0000gp/T/build8264716523375188481.tmp/libraries/StandardCplusplus-master/complex.cpp.o 
In file included from /Users/me/Documents/Arduino/libraries/StandardCplusplus-master/ostream:28:0,
                 from /Users/me/Documents/Arduino/libraries/StandardCplusplus-master/istream:24,
                 from /Users/me/Documents/Arduino/libraries/StandardCplusplus-master/complex:20,
                 from /Users/me/Documents/Arduino/libraries/StandardCplusplus-master/complex.cpp:19:
/Users/me/Documents/Arduino/libraries/StandardCplusplus-master/ostream_helpers: In static member function 'static void std::__ostream_printout<traits, char, float>::printout(std::basic_ostream<char, traits>&, float)':
/Users/me/Documents/Arduino/libraries/StandardCplusplus-master/ostream_helpers:250:109: error: there are no arguments to 'dtostrf' that depend on a template parameter, so a declaration of 'dtostrf' must be available [-fpermissive]
    length = strlen(dtostrf(f, static_cast<int>(stream.width()), static_cast<int>(stream.precision()), buffer));
                                                                                                             ^
/Users/me/Documents/Arduino/libraries/StandardCplusplus-master/ostream_helpers:250:109: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
```

funvill, 2015-11-06:
I get the same errors when compiling with 

ESP8266 Arduino library version: 1.6.5-947-g39819F0
Arduno 1.6.6

krzychb, 2016-03-06:
@probonopd 
@funvill 

I have been looking for `std::sort` and discovered that it is available for esp8266/Arduino out of the box (without installing any additional library)! 

Work done by @igrr and the team never cease to amaze me :+1:

I am using stable 2.1.0 release of 28-Feb-2016 - http://esp8266.github.io/Arduino/versions/2.1.0/ with Arduino IDE 1.6.7. If I switch board from ESP8266 to Adriano UNO, IDE does not find the includes.

Here is my sketch:
```
#include <algorithm>
#include <array>

void setup()
{
  std::array<int, 10> s = {45, 37, 54, 62, 8, 96, 81, 79, 20, 13};

  Serial.begin(74880);
  Serial.println("TempateSorting.ino");

  for (int i : s)
  {
    Serial.print(i);
    Serial.print(" ");
  }
  Serial.println();

  std::sort(s.begin(), s.end());

  for (int i : s)
  {
    Serial.print(i);
    Serial.print(" ");
  }
  Serial.println();
}

void loop()
{
}
```
The result is as follows:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
TempateSorting.ino
45 37 54 62 8 96 81 79 20 13 
8 13 20 37 45 54 62 79 81 96 
```

This is awesome :smile:

Krzysztof

maniacbug, 2023-07-05:
Should be fixed now in 2756386

closed 2023-07-05 by maniacbug