maniacbug/StandardCplusplus #19: Development status of this library

matthijskooijman, 2016-05-18:
I was wondering what the status of this library is. I'd very much like to use this kind of stuff in my Arduino projects, especially the newer C++11 template stuff (so I was happy to find `initializer_list` was added). However, it seems that development is stalled / diverged a bit, so I'm wondering a few things. These questions are as much for @maniacbug, who I think initially created this library, as well as for @kibergus and @rpavlik, who have been making some significant changes in their forks AFAICS.

 - Are there any plans to update to a newer upstream version? Seems this repo contains uClibc++ 0.2.2, while there has been a 0.2.4 release in 2012, and some more activity in the git repo (most recently two months ago).
 - Are there any plans to submit changes back upstream? It seems there are some changes in this repository, and even more in the repositories of @kibergus and @rpavlik that could be useful for upstream uClibc++ as well, while at the same time reducing the maintenance burden for this Arduino port.
 - Are there any plans to merge the ports of @kibergus and @rpavlik into this one, or vice versa? Having a single canonical "latest version", as well as a single place to focus development effort might help in moving forward?

kibergus, 2016-05-18:
Hello. I have no plans on continuing development of this library because I've moved to STM32 from arduino, so I can use usual STL which comes with gcc. So I don't have plans to spend time required to commit my changes back, but I will be glad if anybody will do that.

rpavlik, 2016-05-20:
I haven't worked on arduino projects in a while, and I actually switched to
using stlport instead of this project (too many differences from the
standard to get Boost to work successfully), but if anyone is interested
they're welcome to submit my changes upstream, etc. It does seem like there
continues to be interest in a C++ standard library implementation suitable
for avr, and while this eventually didn't fit my needs, it might fit other
arduino users with simpler requests.

mbrgm, 2016-06-07:
I'm planning on forking the repo, merging my changes (for ESP8266 compatibility), merging @ivankravets changes (for platformio support) and then pointing the platformio-registered library at my fork. So if you have any PRs you want to merge, I could pull them in. But I won't promise extensive maintenance, as I'm also using relatively small parts for fun-/hobby-only projects. However, accepting PRs in a platformio-registered repo will already be better than the current state.

@maniacbug What do you think? Do you have any future plans regarding this project or do you want to abandon it?

maniacbug, 2023-07-05:
I did return back to this and test/merge in the major changes, and address the major problems. As with others I mostly long ago moved onto more powerful silicon. But I will work on keeping this up to date for anyone who's interested.

matthijskooijman, 2023-07-06:
@maniacbug Thanks for all your recent work and keeping this alive!