maniacbug/StandardCplusplus #24: Problem with vector library

TheZeldaBoss, 2017-03-13:
Hello,

I'm currently trying to use your library to make a school project, and I prefer to use vectors instead of double-sized array. It worked well on my last project (made last week), but now, when I try to add the vector library, I get the following errors :

C:\Users\benoit\Documents\Arduino\libraries\StandardCplusplus/vector:43:94: error: macro "swap" passed 4 arguments, but takes just 2

  template <class T, class Allocator> void swap(vector<T,Allocator>& x, vector<T,Allocator>& y);

                                                                                              ^

C:\Users\benoit\Documents\Arduino\libraries\StandardCplusplus/vector:520:106: error: macro "swap" passed 4 arguments, but takes just 2

  template <class T, class Allocator> _UCXXEXPORT void swap(vector<T,Allocator>& x, vector<T,Allocator>& y){

Il works well with the example file and my previous project, so I don't know what to do...

I just include the standardCplusplus.h and vector headers at beginning of my .ino file.

mike-matera, 2017-03-13:
This project is dead. I have a library that's available in the Arduino Library Manager called ArduinoSTL.

TheZeldaBoss, 2017-03-13:
OK, I'll check that, thanks and sorry for disturbing you

maniacbug, 2023-07-05:
Also, feel free to post the entire failing .INO sketch here.

closed 2023-07-05 by maniacbug