maniacbug/StandardCplusplus #27: Problems when compiling library in Arduino IDE

morechicken, 2017-12-11:
It seems like there are some compilation problems related to the istream and ostream classes. I don't think it is related to my code. I get the following errors just after the compiler says the following: 

Compiling library "StandardCplusplus"

In file included from C:\Users\XX\Documents\Arduino\libraries\StandardCplusplus/ostream:28:0,

                 from C:\Users\XX\Documents\Arduino\libraries\StandardCplusplus/istream:24,

                 from C:\Users\XX\Documents\Arduino\libraries\StandardCplusplus/complex:20,

                 from C:\Users\XX\Documents\Arduino\libraries\StandardCplusplus\complex.cpp:19:

C:\Users\XX\Documents\Arduino\libraries\StandardCplusplus/ostream_helpers: In static member function 'static void std::__ostream_printout<traits, char, float>::printout(std::basic_ostream<char, traits>&, float)':

C:\Users\XX\Documents\Arduino\libraries\StandardCplusplus/ostream_helpers:250:109: error: there are no arguments to 'dtostrf' that depend on a template parameter, so a declaration of 'dtostrf' must be available [-fpermissive]

    length = strlen(dtostrf(f, static_cast<int>(stream.width()), static_cast<int>(stream.precision()), buffer));

mike-matera, 2018-04-10:
This project is dead. You can use my library in its place: 

https://github.com/mike-matera/ArduinoSTL

MRES1, 2019-10-27:
Hi Mike thanks for your work, I wanna get to work "Eigen" for Arduino using your library, is this possible?

tonk777, 2020-04-07:
The project may be dead but the fix is easy

You can fix this doing the following edits to remove the duplicated default template argument (removal of = char_traits<charT> )

File: istream
Line: 343
Before: template <class charT,class traits = char_traits<charT> > class _UCXXEXPORT basic_istream<charT,traits>::sentry {

After: template <class charT,class traits > class _UCXXEXPORT basic_istream<charT,traits>::sentry {

File: ostream
Line: 317
Before: template <class charT,class traits = char_traits<charT> >
After: template <class charT,class traits>

Guchshenskaya, 2021-01-11:
> The project may be dead but the fix is easy (its in the pull request #20)
> 
> You can fix this doing the following edits to remove the unnecessary duplicated default template argument on the nested templates (removal of `= char_traits<charT>` )
> 
> File: istream
> Line: 343
> Before:` template <class charT,class traits = char_traits<charT> > class _UCXXEXPORT basic_istream<charT,traits>::sentry {`
> 
> After: `template <class charT,class traits > class _UCXXEXPORT basic_istream<charT,traits>::sentry {`
> 
> File: ostream
> Line: 317
> Before: `template <class charT,class traits = char_traits<charT> >`
> After: `template <class charT,class traits>`

It helps! Thank you!!!

xavier2910, 2021-03-03:
> The project may be dead but the fix is easy (its in the pull request #20)
>
> You can fix this doing the following edits to remove the unnecessary duplicated default template argument on the nested [...]

I was having the same problem, thanks!

maniacbug, 2023-07-05:
Should be fixed now in 9d8277e

closed 2023-07-05 by maniacbug