maniacbug/StandardCplusplus #32: Dead project?

domino2, 2021-04-09:
Last commit on 22 Sep 2013 and it is doesn't work on Platformio and Arduino UNO. I'm getting plenty errors when doing build. But works with the ArduinoSTL lib.

The errors:
```
.pio/libdeps/uno/StandardCplusplus/ostream:318:50: error: default argument for template parameter for class enclosing 'class std::basic_ostream<charT, traits>::sentry'
.pio/libdeps/uno/StandardCplusplus/istream:343:107: error: default argument for template parameter for class enclosing 'class std::basic_istream<charT, traits>::sentry'
.pio/libdeps/uno/StandardCplusplus/ostream:318:50: error: default argument for template parameter for class enclosing 'class std::basic_ostream<charT, traits>::sentry'
```

chen-lee, 2022-01-02:
Just find the error location of the code and then delete '= char_traits<charT>', i tried this, it works.
[image]

maniacbug, 2023-07-05:
Should be fixed now in commit 9d8277e

closed 2023-07-05 by maniacbug