maniacbug/StandardCplusplus #36: Bug when using the library

nilsonLazarin, 2023-10-17:
I'm trying to use the library, but I'm facing this bug.

[image]

My OS is Debian 12
```sh
nilson@acerAspire:~$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
```

My Arduino IDE is 1.8.19
```sh
nilson@acerAspire:~$ apt list arduino
Listing... Pronto
arduino/stable,now 2:1.8.19+dfsg1-1 amd64 [installed]
```

matthijskooijman, 2023-10-18:
Hm, I remember this symptom, IIRC it was because this library ended up including the Arduino core `new.h` header or something rather than the StandardCplusplus version, and the former does not define `_UCXXEXPORT`.

The same thing happened in ArduinoSTL. See https://github.com/mike-matera/ArduinoSTL/issues/56#issuecomment-694429021 for some background, and https://github.com/mike-matera/ArduinoSTL/pull/82 for my proposed (but unmerged) fix for ArduinoSTL (which could probably be adapted to StandardCpluplus as well, but I won't have time for that).

nilsonLazarin, 2023-10-18:
Thanks, @matthijskooijman 

Because I don't need this resource (new) in my project, I only deleted the file new_handler.cpp from library folder and work very well.

[image]